dokutan/rgb_keyboard #1: Problems with Keymapping

keinLeerzeichen, 2020-02-20:
I noticed that, whenever i try to remap a key who's name is longer than a tab (i.e. "Caps_Lock") the keymap ends up remapping my whole keyboard (spacebar becomes "m" and so on).

dokutan, 2020-02-21:
I have added a new branch iso-keymapping that should hopefully address that issue. It would be great if you could test it and reply back with the results. If it fully works (remapping of the Int_Key (<,>,|, etc.) and profiles) i will merge the changes to master.

keinLeerzeichen, 2020-02-21:
Hi, sadly, the problem persists (unless downloading the new branch and using the same commands was wrong^^). The whole keymap is still shifted, tell me if i can help you with testing in any way.

dokutan, 2020-02-21:
Dumb question, did you run make clean befor make upgrade? Also try running the new binary directly from where it was compiled, maybe you were still somehow using the old version.

What i did: I used the ISO version of the official software, and captured the usb communication, this was different from the ANSI version, but it still worked on the ANSI keyboard. Also when remapping the Int_Key it showed up correctly in the official software, even though the key doesn't eyxist on my keyboard.

If you want to help out: You could set up the official software in a Windows VM (QEMU or VirtualBox) with usb passthrough and capture the usb communication with wireshark. Feel free to ask if you need any help, i will add some information about the protocol to the doc directory in master, so look there and in the sourcecode for information. Some links for getting started:
[https://blog.davehylands.com/capturing-usb-serial-using-wireshark/](https://blog.davehylands.com/capturing-usb-serial-using-wireshark/)
[https://wiki.wireshark.org/CaptureSetup/USB](https://wiki.wireshark.org/CaptureSetup/USB)

keinLeerzeichen, 2020-02-21:
Since im very new to git and linux in general i actually removed the whole rgb_keyboard folder and cloned the new one using (im using ubuntu) : git clone -b ISO-keymap ttps://github.com/dokutan/rgb_keyboard.
So i think i should have had to use the correct version.
I have a Windows desktop could i do the wireshark thing from that instead of using a VM?

dokutan, 2020-02-21:
I don't know if wireshark is able to do usb sniffing on windows, from a quick search online it appears to be the case, [maybe this helps](https://geekthis.net/post/usb-sniffing-and-programming/), unfortunately that is something i can't really say anything about, because i don't have a windows pc to try it out.
The git command you provided, does not work, the branch name is wrong and the h from https is missing, i am assuming you didn't make these mistakes when cloning. Just in case the working version is:
``
 git clone https://github.com/dokutan/rgb_keyboard -b iso-keymapping
``
To check wether the version is correct, look inside the include/ directory, only in the iso-keymapping branch a file named data_keymap exists.

keinLeerzeichen, 2020-02-21:
Ok i did clone correctly. I'll look into the Wireshark stuff and do that if I understand it otherwise you'll hear from me again.

dokutan, 2020-02-21:
That's good to hear, if you have any remaining questions, particularly to what you are looking for in wireshark, feel free to ask.

keinLeerzeichen, 2020-02-22:
I set up a VM using qemu and KVM now because i have found a guide that explains that in extreme detail. But now i have trouble bridging my network connection to my VM.

dokutan, 2020-02-22:
The QEMU wiki should help, look in the Guest Hints section: [https://wiki.qemu.org/Documentation/Networking](https://wiki.qemu.org/Documentation/Networking).
If everything fails but you are able to install windows (maybe 7) without a network connection, you can downlaod the installer for the keyboard software on linux and transfer it to the VM via an .iso, or by mounting the VM disk image with guestmount.

keinLeerzeichen, 2020-02-23:
My Win10 in VM now has an internet connection 👍. And i now have wireshark running on my Windows VM with USBPcap. I am not completely sure what data you need exactly, so could you maybe give an example of how it would look like? thanks

dokutan, 2020-02-24:
Im sorry, i guess caused a slight misunderstanding, the purpose of the VM is that you are able to run wireshark on the linux host, which is much easier.
Have you got the usb passthrough of the keyboard working?
If yes, try loading the usbmon kernel module and capturing some usb communication between the keyboard and rhe VM, (look at the wireshark display filter, so you don't see the communication to other devices).
As soon as i am back home i will add some captured data as an example to the doc directory in the master branch.

dokutan, 2020-02-25:
I have added a generic description of the capture process and a screenshot to doc/README.md in master. In this particular case when applying a keymapping, there will be more than one data packet, i encountered 8 or 11 (in addition, start and end).
What you need to capture is the data for the default keymapping. You can export the data by selecting all interesting packets and the choosing File>Export Packet Dissections>As "C" Arrays and choosing selected packets only.

dokutan, 2021-04-17:
I'm closing this issue as it seems to be inactive and partially resolved.

I confirmed that keymapping works on an Tecware Phantom TKL ISO keyboard, so the problem seems to be limited to the 60% GMMK and Ajazz AK33. Furthermore this issue is noted in the README and when remapping the keys a warning is shown and the user gets asked for confirmation.

If someone has a keyboard that is affected by this problem and is interested in providing the captured data, please open a new issue specific to to that keyboard.

closed 2021-04-17 by dokutan