dokutan/rgb_keyboard #2: Possible "instant color mode" ?

diogotr7, 2020-03-17:
Hello, cool project!

I decided to make an issue because i found that my keyboard (Ajazz AK33, pid 0x7903 ) has a pretty similar (maybe even the same) firmware to the ones you're targeting. i wrote a [library + gui](https://github.com/diogotr7/AjazzAK33.NET) to control it that works reasonably well. I will probably grab a couple of the things you're doing that seem to work a little better from very short tests :) .

I was also wondering if you know anything about setting lighting non-permanenly, like the corsair / logitech / razer keyboards allow you to do. The end goal was to add these boards to [Aurora](https://github.com/antonpup/Aurora) for game integration and such but that isnt possible if we're always writing colors to the flash (also, how long does the your program take to set a custom pattern to all the keys at once? I am using windows and haven't had a chance to test it yet)

I will try to get this running on my arch install and will get back to you if it indeed works the same as my tool.

dokutan, 2020-03-20:
Thanks a lot for the feedback.
It would indeed be great if you could report the results from your compatibility tests.

As far as setting the lighting non-permanently, unfortunately i don't see a possibility, unless i missed something, as the official software does not support it.

At the moment it takes quite a while to set a complete custom pattern, this is because i set each key individually. There is a way to do this much faster, however i have not currently implemented that (i might do that if there turns out to be an interest in it). Another problem would be, that the kernel driver gets detached during the writing of any settings, so no input is possible during this period (again if there is interest that could probably be changed).

diogotr7, 2020-03-20:
unfortunately none of the commands i tried had any effect. i of course changed the PID in `rgb_keyboard.h` but could not see any change in the keyboard. 

This command from the readme `rgb_keyboard --brightness 5 --speed 2 --color 00ff00 --leds rain`
takes 13 seconds to finish (as you said, no input is possible during this time). `rgb_keyboard -q` takes around 3 seconds but also has no effect. Then, the program seems to exit gracefully without printing anything. I ran it as root as i didnt setup the udev rules. could that cause any issues? 
My ui works fine under both windows 10 and arch. I grabbed the values from your `write_mode` function from `writers.cpp` and that seemed to fix some of the modes that weren't being applied properly with my tool, so it's very strange that it doesn't work. I admit i don't understand the hid part completely as in c# it's all behind the HidSharp wrapper. maybe the interfaces / endpoints are different?

dokutan, 2020-03-20:
From the lsusb output the keyboards look indeed extremely similar, however your keyboard seems to lack an endpoint to receive data, which seems very strange to me.

The duration these commands take means that the individual transfers timed out.

Could you capture the communication between your keyboard and either the official or your software when setting the mode or brightness with wireshark?
I am suspecting there is a minor difference in the USB endpoints (or maybe the data), so that should help to clear things up. If you need more information, i put a description of the general process in doc/README.md.

Also please try to use the latest version from git for further tests, i changed the interface and some other things today.
You can also open the keyboard by specifying the bus and device (no changing the PID in the source is neccessary).

diogotr7, 2020-03-20:
[ajazzak33-pcap.zip](https://github.com/dokutan/rgb_keyboard/files/4362122/ajazzak33-pcap.zip)

I made some tests with the official app and with my app(those ones say `my_program`). the official one seems less stable than when i last tested it... I wonder why that's the case. I captured these on windows 10 with usbpcap + wireshark. 

I will give the program another try by specifying the bus and device.

let me know if you need any further information, will update when i test again.

dokutan, 2020-03-21:
Thanks a lot for these captures, they really helped.

The only difference seems to be that the packets going to the ajazzak33 are transmitted with an usb control transfer instead of an interrupt transfer.

I have added a new branch ajazzak33 in which i adapted the writing functions for the brightness and mode. If this works, it should be quite easy to adapt the other functions as well to be compatible with both types of keyboard.

If this turns out not to work, capturing the (broken) communication of this software would be neccessary.

diogotr7, 2020-03-21:
i can confirm that writing brightness and mode now works. brightness is 0-5 instead of 0-9 like you say in the readme though which i find a bit strange. anything above 5 looks about the same brightness as if i set it to 1.

dokutan, 2020-03-21:
In the ajazzak33 branch, i have added support for all writers.
You need to use the --ajazzak33 option to enable this. With this option the program opens the keyboard by the PUD 0x7903, so no bus/device specification should be neccessary.

The maximum brightness just seems so be another difference, i will change the appropriate perts accordingly.

If everything works, i will merge these changes to the master branch, after the small differences have been properly documented.

diogotr7, 2020-03-21:
had me scared there, the remap function does indeed fuck everything up. literally all my keys were swapped around. thankfully i had the firmware backed up with the official windows app before i started or i probably wouldn't be typing this right now :D

in any case, some of the combinations of options seem to not want to work together. for example, i couldn't make `-v` do anything. if i set it to the reactive color mode it always defaults to the yellow preset. Is there anything specific you'd like me to try? fixed color works correctly. i didn't try report rates or macros or anything like that after what the remap did. I don't know if is should try custom. my app does custom but one of the keys isn't settable (Fn) and it takes 6+ seconds to set the whole board....

diogotr7, 2020-03-21:
[ajazz-ak33-firmware.zip](https://github.com/dokutan/rgb_keyboard/files/4363135/ajazz-ak33-firmware.zip)

Just in case someone comes across this issue, here is the firmware file i used with the official app.

dokutan, 2020-03-21:
I'm glad it worked out ok.

Everything LED related should probably be safe, however i can not make any promises.
If the basic LED settings (mode, speed, brightness, direction) work, that would be a first step.
As far as custom patterns go, you could try with ``--custom-keys`` option to change individual keys.
To prevent any further problems i would closely compare the features of this to the official software.

To fix the ``-v`` bug, you would have to capture setting the variant from a working software. You could either send me the captured data or try to fix it yourself and make a pull request.

diogotr7, 2020-03-21:
mode, brightness, direction and speed seem to work fine.

Setting a custom pattern worked, but the resulting colors seemed a bit weird. maybe the mappings are different? here's a picture after applying example.conf: 
[image]

I will fight with the official program again to see if i can capture the variant. Might be useful or not, but i got some of the information from this repository [here](https://github.com/thanks4opensource/NotLinuxAjazzAK33RGB)

dokutan, 2020-03-21:
What exactly do you mean is weird about the colors? The default color is white.

- Does setting a key
  - to ff0000 make it red ?
  - to 00ff00 make it green ?
  - to 0000ff make it blue ?
- Can all keys be set?
This should answer if either the key offsets or packet structure is at fault.

Unfortunately i have some trouble completely understanding the led part of the code from your linked repository.

dokutan, 2020-03-28:
I am closing this issue for now, as the Ajazz AK33 is now supported in the master branch.
If there is still interest in integration with other software or information regarding the missing features of the Ajazz AK33 support, please open a new issue for that purpose.

closed 2020-03-28 by dokutan