dokutan/rgb_keyboard #5: Redragon YAMA K550 support

slightlyfaulty, 2020-09-27:
I have a Redragon YAMA K550 keyboard. Unfortunately this doesn't seem to work with it at all.

Is there any testing/debugging I can do, or output I can post, to help get it working? Would be amazing to be able to control the lighting for this kb from Linux.

I'm running Arch if it makes any difference.

dokutan, 2020-09-27:
The output of ``lsusb``, ``lsusb -d vid:pid`` the error message of the program if there is one would be helpful.
Try running with ``sudo``, the ``--bus`` and ``--device`` options and the ``-A`` option. Try ``rgb_keyboard -r`` and the options.
You could check if any data is send to the keyboard with wireshark (setup is similar to https://github.com/dokutan/rgb_keyboard/blob/master/doc/README.md#capturing-usb-data). If data is sent but the keyboard ignores it you would have to install the official software and capture data.

slightlyfaulty, 2020-10-01:
Thanks for the info. I tried your suggestions, and it seems to be working! `--bus`, `--device`, and `-A` are all needed. Without `--bus` and `--device`, this message is shown:

```
Could not open keyboard, check hardware and permissions.
Try with or without the --kernel-driver option.
```

With `--bus` and `--device` but without `-A`, nothing happens and no output is shown. Toggling `sudo` or `--kernel-driver` seems to make no difference in either case.

This is the device output from `lsusb`:

```
Bus 003 Device 006: ID 0c45:5204 Microdia USB DEVICE
```

Thank you for making this great tool! From reviews I've read, not even Windows users have this much control over this keyboard using the official software.

dokutan, 2020-10-02:
I'm glad it that it worked for you, i have updated the readme with the findings. There is just one question this creates for me, have you tried the release version or the master branch? In the master branch the keyboard should be detected, but the fact that you had to use the ``--bus`` and ``--device`` options makes me wonder if there is a bug in the detection.

> Thank you for making this great tool! From reviews I've read, not even Windows users have this much control over this keyboard using the official software.

Thanks for that and your issue, contributions like this are what makes it possible to support multiple keyboards in the first place.

slightlyfaulty, 2020-10-08:
Sorry for the delayed reply. I followed the instructions in the readme and compiled from master. It just gives me that error no matter what I try, if I don't specify `--bus` and `--device`.

dokutan, 2020-10-08:
Thanks, that information was really helpful. There was indeed a bug in the handling of multiple USB product ids, that should be fixed with the latest commit (i should rewrite this part of the code when i find the time to do so). Feel free to close this issue if it works, if it still doesn't work or some other features of the software don't work on your keyboard please let me know.

slightlyfaulty, 2020-10-08:
Awesome. It seems to be working now without `--bus` and `--device`.

One thing that I can't get working properly is loading a custom pattern file. Most keys change to the specified color, but other keys just refuse to change and don't light up at all.

This is the result of loading your `example.conf` but with all colors set to the same coral pink color:

[image]

Strangely, online resources say that profiles on this kb can be reset by holding down the `LR` button for 5 secs, and that you can change the color of individual keys in `LR` mode, but neither of these seem to work for me at all. I'm not sure if your tool has something to do with it, or if I have a faulty keyboard, or what... 😕

dokutan, 2020-10-08:
Could you install the official software in a VM or a Windows machine and capture the USB communication when you individually change the color of the affected keys (keys that don't work or have the wrong label in the software)? That would be the only way to fix this issue. You can find further information on that process in doc/README.md or ask if anything is unclear.

slightlyfaulty, 2020-10-08:
I have a VMware Windows guest set up on my Linux host, but I can't seem to get the keyboard software to detect my kb. Do you know of any specific trick to get this working?

In any case, it's not that big of a deal. At least I can set my kb to a solid color of my choosing with your software which was not possible before. Thank again for sharing this great tool.

dokutan, 2020-10-08:
I have no experience with VMware, but this sounds like a problem with missing USB passthrough. Does Windows recognize the keyboard as a USB device in the device manager (it should), can the Linux host receive input events from the keyboard when the VM is running (it shouldn't)?

If this problem doesn't matter to you and you don't want to continue work on it, that is fine for me. Please let me know though, so i can add this as a more permanent bug to the README.

slightlyfaulty, 2020-10-08:
Yeah, gonna have to put a pin in it for now 😄

Thank for all of your help so far. Will leave it with you if you wanna close this issue. I'm happy enough with everything that is working on this keyboard now.

Keep up the excellent work!

dokutan, 2020-10-08:
No problem, i will close this issue as the original issue has been resolved. If you, or someone else who is reading this finds the time and interest to capture data and work on the broken custom key colors, just open a new issue for that purpose.

And thanks to you for opening this issue and getting it resolved, without this there would be no support for this keyboard.

closed 2020-10-08 by dokutan