dokutan/rgb_keyboard #6: Redragon K556 Devarajas support

ionsquare, 2020-12-05:
Hey, this is a really cool project.

It's somewhat working with my K556 Devarajas (`--ajazzak33` required to work), I'm able to set brightness, speed, and colours and trigger the preset patters, but I'm unable to set specific key colours with `--custom-keys` or set use `--custom-pattern`.

Using `--custom-keys` does nothing at all, and using `--custom-pattern` sets a really weird preset, always the same regardless of what input file I use. For example, using `/dev/null` as the input file still gives the same result 

```bash
rgb_keyboard --ajazzak33 --custom-pattern /dev/null
```
[image]

Any suggestions for how I could troubleshoot this to get it working?

ionsquare, 2020-12-05:
Oh never mind, it must have somehow remembered the previous settings file, I changed the colours in the file (e.g. red to green) and when I set `--custom-pattern` again those red keys did turn to green.

dokutan, 2020-12-05:
I'm glad that changing the colours worked, however the keys aren't accessible under their actual name in the file, e.g. ``x`` is not called ``x`` but some modifier.

To fix that you could try to change each key individually and create a list that maps the name of the key to the actual key. If you can't change the colours of some keys with this method, you would have to set up the official software on a Windows machine (or VM) and capture the usb traffic when changing the colour of the keys with wireshark.

With that information it should be relatively easy to add support in this software.

kanguru007, 2021-06-01:
How can I find the correct values for this table? The 'I' is actually changing the 'Q'. The values from Linux showkey are also wrong.

const std::map<std::string, std::array<uint8_t, 3>> rgb_keyboard::keyboard::keycodes_ansi = {
	{"Esc", {0x57, 0x03, 0x00}},
	{"F1", {0x5a, 0x06, 0x00}},
	{"F2", {0x5d, 0x09, 0x00}},
	{"F3", {0x60, 0x0c, 0x00}},
(...)

dokutan, 2021-06-01:
These values were obtained by capturing the data sent by the official software with wireshark, while individually changing the color of all keys, then comparing the data to find the differences between the keys. Alternatively you could use systematic trial and error if they are simply swapped.