dokutan/rgb_keyboard #7: Redragon K587 PRO Magic Wand

HeuFui, 2020-12-21:
I don't know if this is the right place to leave this message, but I'm just here to report that this program works on this keyboard. It needs the --ajazzak33 option, though.

I'm running it on Arch Linux. No issues found.

dokutan, 2020-12-21:
Thank you for this info, this is indeed the right place. I have updated the README to include this keyboard. Could you post the output of ``lsusb`` or the USB Product ID?

I am asking because these features are known to not work on many keyboards, have you tried a custom LED pattern or remapping the keys? Please be careful should you try remapping the keys, i recommend having the official software installed somewhere in case all keys get shifted around.

HeuFui, 2020-12-21:
The output of lsusb:

`Bus 001 Device 003: ID 0c45:5004 Microdia Redragon Mitra RGB Keyboard`

> have you tried a custom LED pattern or remapping the keys?

No, not yet. Nor I have thought of doing so. However, if I could create a custom LED pattern, that would be great. Maybe an experience for another time. ;)

> Please be careful should you try remapping the keys, i recommend having the official software installed somewhere in case all keys get shifted around.

Thanks for this warning. I'll be sure to have the software around to correct some blunder if I happen to try that later on.

dokutan, 2020-12-21:
Thanks, you can edit [examples/example.conf](https://github.com/dokutan/rgb_keyboard/blob/master/examples/example.conf) and apply it with ``rgb_keyboard -l custom -P example.conf`` to get a custom LED pattern. I am leaving this issue open for now, so feel free to report your results whenever you try it out.

HeuFui, 2020-12-22:
I've tried the custom pattern, using the example.conf file you indicated. Well, some kind of pattern emerges, but the results are not consistent with the input. Some keys are not lit, some are lit in wrong color. 

For example (taken from example.conf):
```
Esc	ff1111 (not lit)
F1	888888 (white)
F2	888888 (white)
F3	888888 (white)
F4	888888 (white)
F5	ff1111 (red)
F6	ff1111 (red)
F7	ff1111 (red)
F8	ff1111 (red)
F9	888888 (white)
F10	888888 (white)
F11	888888 (white)
F12	888888 (white)
PrtSc	ff1111 (not lit)
ScrLk	ff1111 (not lit)
Pause	ff1111 (not lit)
```
The same happens with the rest of the keyboard.

dokutan, 2020-12-22:
This problem is not unique to this keyboard, you can see a photo of the resulting pattern in issue #6, i assume it looks identical on your keyboard.

The best method to fix this would be to install the official software (either in a VM or on bare metal) and use Wireshark to capture the usb communication while changing the color of each key individually (thus creating a mapping between the key and the sent data). If that is something you are interested in, [here](https://github.com/dokutan/rgb_keyboard/tree/master/doc#capturing-usb-data) is a basic explanation of the process, and feel free to ask any questions.

HeuFui, 2020-12-22:
> This problem is not unique to this keyboard, you can see a photo of the resulting pattern in issue #6, i assume it looks identical on your keyboard.

It's a very similar pattern indeed!

> The best method to fix this would be to install the official software (either in a VM or on bare metal) and use Wireshark to capture the usb communication while changing the color of each key [...]

Since I don't have a Windows machine (nor a configured VM), I'll ask a friend to try that out. The learning curve seems steep, but having the payoff of knowing how to proper map this keyboard with your software may inspire me to do it. Thanks for pointing that out!

HeuFui, 2020-12-23:
I've added a simple script with a gui to select the profile you want to use. It's one of the most useful features of rgb_keyboard, as my particular model only has one customizable led pattern. Now I have 3!

Said script:
https://github.com/HeuFui/Scripts/blob/main/rgb_keyboard_profile_selector_gui.sh

dokutan, 2020-12-23:
That  looks quite useful, and reminds to publish the script to change the color i wrote some time ago: [here it is](https://github.com/dokutan/scripts/blob/main/src/keyboard-color.sh).

Maybe i should have another look at the existing [GUI](https://github.com/dokutan/rgb_keyboard#gui).

dokutan, 2021-04-17:
I'm closing this as the original purpose of this issue has been fulfilled. For further discussion about the broken support for custom LED patterns see issue #6.

closed 2021-04-17 by dokutan