dokutan/rgb_keyboard #8: Redragon K552RGB-2

abduct, 2020-12-26:
From what I can tell all the features are working with the -A option supplied for this keyboard.

Only two issues I can see is perhaps a color code representation issue. For example FFFFFF is more of a yellow gold rather than a white. Second issue is after a reboot the keyboard turns off after a few seconds of inactivity. I deleted the udev rule, rebooted, and unplugged/replugged the keyboard and it resumed to work as normal. Not sure if this was a bug due to my linux install or the software.

I have not tested key mapping or macros on this keyboard because of the advisories of them being broken.

Is there a testing process that one can go through to check which features the keyboard supports etc.

Thanks

dokutan, 2020-12-26:
To test whether your keyboard supports key mapping, have the official software and a second keyboard available and try it out. The problem on some keyboards has been that all keys are shifted around. Macros are not working at all.
To fix the key mapping, capture the data sent by the official software when remapping with wireshark, to implement macros capture the data and understand it.

dokutan, 2021-04-17:
Closing this as all your questions seem to be answered, please open a new issue if there is anything else.

closed 2021-04-17 by dokutan