dokutan/rgb_keyboard #9: Running the example.conf flickers some keyboard leds and never actually changes the colors

SushyDev, 2021-01-14:
I tried running some other commands, running -c 00ff00 makes all keys green and i can change brightness and default color effect

I noticed running --brightness to anything above 4 would disable the keyboard lighting, only 1-4 works 

Keyboard: 2019 GMMK

SushyDev, 2021-01-14:
Figured out it only works on active profile 1

dokutan, 2021-01-14:
You can use the ``-p`` option to change the profile to which the settings are applied, then use the ``-a`` option to change the active profile. Could you post the output of ``lsusb`` and ``rgb_keyboard -r`` to allow me to identify the reason for the brightness issue?

dokutan, 2021-04-05:
I'm closing this due to inactivity and because the main reason for this issue is resolved.

closed 2021-04-05 by dokutan