dokutan/rgb_keyboard #13: Redragon K552RGB-1 ABNT2 (Brazilian Layout) with wrong keycodes

TheJayGatsby, 2021-03-06:
### Problem
Whenever I try to load  a color profile, the wrong keys are affected. For instance, A change in key "5" would affect the key "2", a change in the key "f" would affect the key "delete" and so on...

### Diagnostic
It seems that this keyboard uses a different set of keycodes. I mapped which codes were related to which keys. I also found that there were some keys whose codes were not defined in data.cpp.

### Solution
After discovering the codes for all the keys (good old brute force), I changed the data.cpp file and recompiled. It worked perfectly!

[data.cpp.gz](https://github.com/dokutan/rgb_keyboard/files/6095973/data.cpp.gz)

dokutan, 2021-03-06:
I included the data and a new ``--layout`` option, running ``rgb_keyboard --layout brazil --custom-pattern keymap`` should hopefully work now.

TheJayGatsby, 2021-03-07:
It works. Thanks. Closing the issue.

closed 2021-03-07 by TheJayGatsby