dokutan/rgb_keyboard #17: Redragon SURARA K582 support

Wallvon, 2021-12-03:
Hi there,

I have a Redragon SURARA K582 keyboard, and I would like to add support for it. I have set up a VM and wireguard, but the issue is that I have no idea what interactions with the software I should actually capture since "start the capture, trigger some action, stop the capture, save if needed, repeat" is not really that helpful.

Output of `lsusb -v -s 001:004` (if needed): https://pastebin.com/8CLK93hH

dokutan, 2021-12-03:
Have you tried using this software? The vendor and product id is identical to the K556 and K587 which makes it likely that the K582 is already compatible.

Wallvon, 2021-12-03:
The software seems to work (I got an error when I first tried it, but I don't anymore), however, it keeps saying that the brightness format is wrong and it seems like the max is 5. Though, vertical-wave and fixed is multi-colour without the checkbox being checked. And, reactive-color also seems to just be the pink version of the "surmount" colour scheme (which also happens to be my default colour scheme).

dokutan, 2021-12-03:
Thanks, i have added the K582 to the list of supported keyboards in the readme.

Some of the issues you described apply to other keyboards as well, however as multiple models share a product id i see no reliable method to detect the correct one and to e.g. enforce a brightness limit of 5 or disable certain colour schemes.

closed 2021-12-03 by dokutan