dokutan/rgb_keyboard #21: GMMK TKL keyboard doesn't work with example commands.

Essasetic, 2022-01-18:
Hey, I wanted to control my RGB with my GMMK TKL keyboard. I can change from the active profiles but doing anything else either completely turns off the RGB on the keyboard or does nothing (running from sudo).

the lsusb output is:
`Bus 001 Device 002: ID 0c45:652f Microdia Backlit Gaming Keyboard`

If I don't run with sudo I get this:
`Could not open keyboard, check hardware and permissions.
Try with or without the --kernel-driver option.`
(the --kernel-driver flag also does nothing).

Any help would be much appreciated.

dokutan, 2022-01-19:
What layout does your keyboard have and what exactly have you tried to control?
I confirmed that this software works with at least one ISO TKL GMMK, but different layouts and firmware versions can be the casue for problems. You could also try using only the lower brightness values, on some keyboars higher values cause the rgb to turn off.

Essasetic, 2022-01-29:
> 

Sorry for the late reply, returned my keycaps and ordered a new set but they haven't come until now. Yeah my layout is ISO TKL.

dokutan, 2022-01-29:
Have you tried brightness values below 3?