dokutan/rgb_keyboard #23: Redragon Draconic

souovan, 2022-04-21:
I don't know if this is the right place to make this question, but here it goes: There is any plan to add support to Redragon Draconic keyboard ?

ps.: if there isn't the place please close this issue.

dokutan, 2022-04-21:
I have generally no plans to add support for more keyboards on my own, because supporting a new keyboard requires physical access to it. This means, if you want support for the Redragon Draconic, you would have to provide:
- at least the USB vendor and product ids
- feedback/testing after features are implemented
- potentially captures of the communication with the official software
I further suggest you look at the closed issues regarding support for new keyboards if you are interested.

souovan, 2022-11-07:
Thank you for you feedback!

closed 2022-11-07 by souovan