dokutan/rgb_keyboard #25: Can read keyboard but result never changes

Razzula, 2022-07-14:
The command ` rgb_keyboard.exe -r --kernel-driver` returns:
```
Active profile: 1

Profile 1:
Led mode: unknown
Direction: right
Color: 11f06f
Brightness: 1735287116
Speed: 1
Report rate: 125 Hz

Profile 2:
Led mode: unknown
Direction: right
Color: fbe96f
Brightness: 50
Speed: 1
Report rate: 125 Hz

Profile 3:
Led mode: horizontal-wave
Direction: right
Color: 10f06f
Brightness: 0
Speed: 3
Report rate: 125 Hz
```
However this result never changes, even if I manually change the active profile, colour values, etc. For example, after changing the Hz to 250, re-running the command still returns `Report rate: 125 Hz`.

None of the commands that change the values (such as changing profile, or key colours, etc.) do anything. No console output.

Using a Phantom Tecware TK ISO keyboard.

dokutan, 2022-07-14:
Interesting that it even compiled on windows, did you have to change anything?
Unfortunately i don't have a windows machine to replicate your setup, a few things you could try for troubleshooting:
- capture the usb communication with wireshark and attach it to this issue, maybe something weird is happening there
- try this program on Linux (optionally in a VM with USB passthrough)
- Check if you have the correct libusb drivers installed or any other limitation applies? (https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows)

Razzula, 2022-07-14:
Yeah, I didn't think it would work at all, but figured I'd try it out. Didn't have to change any of the source code, other than the filepaths in the makefile. Was planning on testing it on Linux when I got the chance, to ensure it wasn't the keyboard itself. I'll let you know once I have, along with the wireshark capture.

I think I used the correct drivers, but will double check.

Thanks :)

Razzula, 2022-07-14:
Just tested it on my Linux machine: worked perfectly. Didn't test it very extensively but `-r` returned correct data and `-a` worked as expected. All these worked without using `--kernel-driver`, by the way, whereas on Windows `-r` insists on it.

Back on Windows, Wireshark doesn't detect anything when running `rgb_keyboard.exe -a 2 --kernel-driver` or `rgb_keyboard.exe -r --kernel-driver
`, but does detect keypresses, and when I change the profile using the official software. And the drivers are correct. So I guess the problem is to do with how libusb works on Windows, and so the signals aren't being sent.

Strange that that `-r` does return a result though, despite the keyboard not sending any data. Does the code have default values for the results or something?

dokutan, 2022-07-14:
> Strange that that -r does return a result though, despite the keyboard not sending any data. Does the code have default values for the results or something?

Yes, default values are used. In this case opening the keyboard succeeds, otherwise an error would be shown.

Razzula, 2022-08-03:
I've just had another look at the problem, and turns out it was a driver issue after all. The keyboard I am using uses two USB interfaces, one to handle keypresses, and the other to handle the RGB. After changing the Interface1 driver from HID to WinUSB the program performs exactly as expected when testing the `-r` and `-a` commands.

(Note: As a consequence of this, however, the official Tecware software no longer works on the keyboard. Though this doesn't bother me, and the driver can be reset incredibly easily.)

closed 2022-08-03 by Razzula