dokutan/rgb_keyboard #27: Redragon K589 Shrapnel compatibility

purplebar0, 2024-01-27:
What works:
- Profile switching, applying to profile
- Mode (pattern), colour, speed, direction assigning
- Brightness values correctly identified (0-5)

Unsure:
- Reactive colour mode variant (how is it supposed to work?)
- Keymapping and key names; haven't tested it
- Report rate; unsure what's the default as control transfer mode was enabled by default

What doesn't work:
- Custom pattern; using the example configuration, big colourless gaps occur

dokutan, 2024-02-08:
Could you please tell me what the USB product id of keyboard is (you can run `lsusb` to show it)? That would allow me to add it to the readme.

> Reactive colour mode variant (how is it supposed to work?)

The `reactive-color` led mode has four differently colored variants that can be set with the `-v` option, but not all keyboards support this feature iirc.

purplebar0, 2024-02-08:
`lsusb` entry:
```
0c45:5004 Microdia Redragon Mitra RGB Keyboard
```

> `reactive-color`
Well, setting the keyboard to that definitely does something, as I type it fades to some other colour and fades to the first one after a delay. It hardly feels special at all as it's hard to notice, so I feel like it isn't working correctly.

dokutan, 2024-02-08:
It seems to be working correctly then, my keyboard behaves the same. In case you are interested in custom patterns, you can have a look at #13 and 7cbf3b274a6e7fcf20934f879730f92962c3fe0d.