dokutan/rgb_keyboard #30: Support for Redragon Fizz k617

imnottimaq, 2024-05-21:
is there any support for this keyboard? if not, will it be added in future?

dokutan, 2024-05-22:
> is there any support for this keyboard?

This keyboard is not explicitly supported, but it might (partially) work if the firmware it uses is compatible with this software, however only someone who has this keyboard can answer that question. You can:
- run `lsusb` to show the USB VID and PID and compare them with the table in the README
- try it out

> will it be added in future?

Maybe, if you (or someone else with the keyboard) provides enough information, i.e. VID, PID, details about the USB protocol, ...

imnottimaq, 2024-08-11:
T:  Bus=03 Lev=01 Prnt=02 Port=04 Cnt=01 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=258a ProdID=0049 Rev=01.01
S:  Manufacturer=BY Tech
S:  Product=Gaming Keyboard
C:  #Ifs= 2 Cfg#= 1 Atr=a0 MxPwr=500mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=   8 Ivl=1ms
I:  If#= 1 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=82(I) Atr=03(Int.) MxPS=  16 Ivl=1ms

hope this information is useful

sivansh11, 2025-01-06:
ID 258a:0049

so I guess no support ?

dokutan, 2025-01-06:
> ID 258a:0049
> 
> so I guess no support ?

It seems very unlikely that the protocol is compatible, i'm closing this issue for now.

closed 2025-01-06 by dokutan