dokutan/rgb_keyboard #31: K551 Mitra support

hirak99, 2024-07-28:
Hi,

I have a K551 which I would love to control with this program.

Physically the K551 model is almost identical to K552 (which is in the supported list), except it also has a Numpad.

I was playing around with the code to check if I can add support for it.

The device reads as -

```sh
$ lsusb
...
Bus 001 Device 002: ID 320f:5000 Evision RGB Keyboard
...
```

Adding `0x5000` to the `keyboard_pid` list made it recognize it, however I was unable to set anything using the binary.

The following gives no results -
```sh
$ ./rgb_keyboard --leds rain --color 0000ff
(Waits for 2s and returns with no error but nothing changes on keyboard)
```

_However_, it does communicate with the keyboard (since I can't type anything for 1 or 2 seconds when it runs unless I use `--interface0`).

The readout appears to have garbage. None of the modes match with what I have set, also Profile2 appears to have bad values for mode and brightness -
```
./rgb_keyboard --read
Active profile: 1

Profile 1:
Led mode: ripple
Direction: right
Color: 000000
Brightness: 0
Speed: 3
Report rate: 125 Hz

Profile 2:
Led mode: unknown
Color: 7c0000
Brightness: 31867
Speed: 1
Report rate: 125 Hz

Profile 3:
Led mode: horizontal-wave
Direction: right
Color: 000000
Brightness: 0
Speed: 3
Report rate: 125 Hz
```

Question: Is there anything I can try to debug or get it working?

Could the Keyboard be using a different protocol for USB communications (I can see defined in `data.cpp` and used in `writers.cpp`), and if so, is there anything I can use to debug and detect what is the USB protocol it uses?

Currently, openrgb works on my keyboard, so it can be done - just need to find out how.

dokutan, 2024-07-29:
The general idea is to use wireshark to capture the USB communication with the official software, then recreate the data that was sent in writers.cpp (expect some trial and error). Considering that openrgb and rgb_keyboard have at least some support, i recommend capturing that data as well and doing a diff.

However, before that i would try to enable the control transfer mode used by some keyboards, e.g. by adding the PID to: https://github.com/dokutan/rgb_keyboard/blob/8bf5e856cf5c311a2bf2c8ac0aa20f546bdf16d1/include/rgb_keyboard.h#L56

hirak99, 2024-07-31:
Thanks! With control transfer predefined patterns like this are working -

```sh
# This works!
./rgb_keyboard --leds fixed --color 70ff00
# And this works!
./rgb_keyboard --leds ripple --color 70ff00 --brightness=2 direction=right
```

However, custom key lighting is not working as expected -
```sh
./rgb_keyboard --leds custom --custom-keys "z=00ff00;"
# Should light up only z, but this happens -
# https://imgur.com/a/23YF7j9
```

Custom per-key lighting also works from openrgb. I actually want to use rgb_keyboard for per-key lighting, with my own configuration files.

Thanks for the pointer to Wireshark - may be I can monitor how openrgb does this with Wireshark. Haven't used it before, seems like a nifty tool to learn.

Is there any other trick I can try before that to see if per-key lighting already works with some other configuration? Looks like `write_custom()` does it, may be I can tinker it to see if any change would make it work.

hirak99, 2024-07-31:
Correction, the keyboard is K552 Kumara, with a second PID, according to [openrgb's registration](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/EVisionKeyboardController/EVisionKeyboardControllerDetect.cpp#L118).

```
...
REGISTER_HID_DETECTOR_IP("EVision Keyboard 320F:5000", DetectEVisionKeyboards,   EVISION_KEYBOARD2_VID, REDRAGON_K552_V2_PID,      1, EVISION_KEYBOARD_USAGE_PAGE);
...
```

hirak99, 2024-07-31:
I used Wireshark on OpenRGB using it, also am looking at the OpenRGB code.

Also have been researching, found this very nice video on EVision chips that power the RGB in these keyboards - https://www.youtube.com/watch?v=2MHtLEiXaXg

dokutan, 2024-08-03:
If this keyboard uses different codes to identify the individual keys for custom patterns (and key remapping), you could add the keycodes in `rgb_keyboard.h` and `data.cpp`, similar to e.g. https://github.com/dokutan/rgb_keyboard/blob/8bf5e856cf5c311a2bf2c8ac0aa20f546bdf16d1/include/data.cpp#L224

and allow them to be selected here: https://github.com/dokutan/rgb_keyboard/blob/8bf5e856cf5c311a2bf2c8ac0aa20f546bdf16d1/rgb_keyboard.cpp#L230

Figuring out the keycodes is possible by reverse engineering the USB communication or by trial and error (changing the values in data.cpp).

amitsandhel, 2024-08-06:
> Thanks! With control transfer predefined patterns like this are working -
> 
> ```shell
> # This works!
> ./rgb_keyboard --leds fixed --color 70ff00
> # And this works!
> ./rgb_keyboard --leds ripple --color 70ff00 --brightness=2 direction=right
> ```
> 
> However, custom key lighting is not working as expected -
> 
> ```shell
> ./rgb_keyboard --leds custom --custom-keys "z=00ff00;"
> # Should light up only z, but this happens -
> # https://imgur.com/a/23YF7j9
> ```
> 
> Custom per-key lighting also works from openrgb. I actually want to use rgb_keyboard for per-key lighting, with my own configuration files.
> 
> Thanks for the pointer to Wireshark - may be I can monitor how openrgb does this with Wireshark. Haven't used it before, seems like a nifty tool to learn.
> 
> Is there any other trick I can try before that to see if per-key lighting already works with some other configuration? Looks like `write_custom()` does it, may be I can tinker it to see if any change would make it work.

@hirak99  so you got your redragon to work with openrgb and can make it use the new patterns and how did you get it working if I may ask >

I have the redragon k582 but It doesn't work with openrgb at all Id like to try it with my redragon k582?

@dokutan  does your software allow us to program our own themes for our redragon keyboards ?

hirak99, 2024-08-06:
@amitsandhel someone already added it to OpenRGB, it works out of the box for k552.

The easiest thing to try is [to add your product id to rgb_keyboard.h here](https://github.com/dokutan/rgb_keyboard/blob/8bf5e856cf5c311a2bf2c8ac0aa20f546bdf16d1/include/rgb_keyboard.h#L45).

You can then compile this project `make clean; make` and test if it works for your keyboard.

If it does, you can send a PR to this project.

You can find the product id typing `lsusb`, it should have a line that reads like this -

```
Bus 001 Device 002: ID 320f:5000 Evision RGB Keyboard
```

In that example 320f is vendor id, 5000 is product id.

You can also do this on openrgb if you want.

@dokutan thank you for acquainting me to Wireshark.

I will send a PR for K552 but it may take some time because (a) I still need to figure out the key codes and (b) because of some bureaucratic reasons - my current company has a process that I need to follow. (It's a bit cumbersome but I since I moved out of California and the moonlighting laws are ambiguous here, probably best for me to try to follow it. There's also a growing chance that I will leave which might make it easier for me to contribute to open source.)

If anyone in the meanwhile solves this, that's great.

With Wireshark, I have also been successful to use it to create a script to set RGB leds for another device (HP X1000 speakers), for which no other way exists in Linux to my knowledge. This was an unexpected and pleasant turn of events!

amitsandhel, 2024-08-06:
> @amitsandhel someone already added it to OpenRGB, it works out of the box for k552.
>
> The easiest thing to try is [to add your product id to rgb_keyboard.h [...]

@hirak99  apologies but I am still confused. 
My question wrt to to openRGB is do redragon keyboards have accesss to directmode, which would then allow us to have other animation effects aside from whatever comes default with the keyboard. 

And does this software do the exact same thing. I can then create my own custom pattern per say and have my keyboard display that pattern somehow ?

hirak99, 2024-08-06:
I'm sorry the fault is mine, I misunderstood your question. 

With OpenRGB I can statically change the colors on a per key basis. 

However I don't think it has a direct mode. The factory firmware stores the colors in memory every time it's set, which causes delay and flicker. So a new animated pattern is not possible with it. For that the firmware needs to be changed, and this video goes deeper into that topic: https://www.youtube.com/watch?v=2MHtLEiXaXg

amitsandhel, 2024-08-07:
> I'm sorry the fault is mine, I misunderstood your question.
>
> With OpenRGB I can statically change the colors on a per key basis.
>
> However I don't think it has a direct mode. The factory firmware [...]

@hirak99  
yes thats my issue as well. I can use openrgb to use the default in memory rgb themes and some per key basis but using direct modei s impossible. i tried changing the firmeare but sonix didn't even detect my keyboard out of the box.